erichelgeson/BlueSCSI-v1 #106: Allow for easier file names Any extension allowed (hda/dsk/xxx)

aihk, 2022-05-02:
Not all formats are supported. The DSK I made with Mini vMac is not recognized properly

erichelgeson, 2022-05-02:
Mini vMac only supports HFS Partitions - a real hard drive has a Partition Map, SCSI Driver, and HFS partition. Mini vMac has said they may support real drive images in the future. 

Note any file name extensions currently works as there are no standards around them, we just normally align with rasci's hda.

You can find more info about supported emulators here https://github.com/erichelgeson/BlueSCSI/wiki/Transferring-files-(Mac)

aihk, 2022-05-02:
Thanks a lot for the warm reply! I will keep following you.

aihk, 2022-05-02:
It's a pity that I can't use Basilisk II to make images with plus support