erichelgeson/BlueSCSI-v1 #136: SCSI Director 4 sense error (vendor sense page 0x39)

erichelgeson, 2022-06-25:
Currently since BlueSCSI is reporting as a `QUANTUM` `FIREBALL1` and SCSI Director 4 tries to issue a vendor specific request for sense page `0x39`. Since this is not handled we return an illegal sense request - which SCSI Director shows to the user. The user can just hit OK to continue and the format completes fine.

We could either:

* Find the vendor documentation on the format of the `0x39` VDR response and send that back.
* Identify as a different drive that SCSI Director does not send VDR requests for, but is still in the Apple approved drives.
* Send back all 0's (this does work, but we dont know the format or what this means, or how long it should be)
* Nothing, people don't really use SCSI Director to format, and if you hit OK it works anyways.

Mostly logging this issue for someone (myself included) in the future to revisit.

erichelgeson, 2022-06-25:
@mynameistroy already found them, thanks! http://bitsavers.org/pdf/quantum/ProDrive/Quantum_ProDrive_40S_80S_Product_Manual_Jun88.pdf

[image]

mactcp, 2022-06-25:
With the addition of the Apple mode sense, I'm finding that Apple's tools are a lot less picky now. HD SC Setup seems to work no matter what the drive identifies as, and Drive Setup only seems to care about the vendor, the product name doesn't seem to matter any more.

For example, I have my BlueSCSIs identifying as a `QUANTUM` `BLUESCSI F1 64GB` and it works with both HD SC Setup and Drive Setup, no patches to those tools needed. Using this configuration, I'm not seeing any errors due to vendor specific requests from SCSI Director.

erichelgeson, 2022-06-26:
Went with the `Identify as a different drive that SCSI Director does not send vendor requests for, but is still in the Apple approved drives.` option :)

closed 2022-06-26 by erichelgeson