erichelgeson/BlueSCSI-v1 #160: jlcpcb issue with BlueSCSI 1.1-a-DB25-Base.stl

Rev-3ng, 2022-08-20:
I tried to order some of these with jlcpcb but they sent back the following issue:

"As shown below, there's a small gap at the indicated position, during ptinting the gap might be expanded and caused crack or deformation for the part.

Will this risk be acceptable for you?"

[image]

erichelgeson, 2022-08-25:
I've never noticed this but yes there is a gap, though when printing I've never seen it. You may want to use Bruce's much better case here https://tinkerdifferent.com/threads/bluescsi-open-low-cost-diy-scsi-to-sd-device.29/page-4#post-5075 

I'll leave this open until i replace this file with Bruces or fix this stl.

lausvi, 2022-09-20:
It can be printed ok, although the gap shows in the bottom piece. I added a pull request for my slightly edited remix of the case (fixes the gap as well).