erichelgeson/BlueSCSI-v1 #171: Add Atari TT030 to list of supported systems

fdanapfel, 2022-09-20:
Hello,

I've successfully used the "Desktop v1.1" version of BlueSCSI with firmware [v1.1-20220626](https://github.com/erichelgeson/BlueSCSI/releases/tag/v1.1-20220626) as a replacement for the internal SCSI disk in an Atari TT030 (with the stock Motorola 68030 32MHz and TOS3.06).

Maybe you could therefore  update the list of [Supported Systems](https://github.com/erichelgeson/BlueSCSI/wiki/Supported-Systems) to also include the Atari TT030 as "Working".

uweseimet, 2022-11-03:
@fdanapfel Can you please provide more information on your setup? When I connect BlueSCSI to my TT the SCSI bus is blocked. Now and the emulated devices are detected, but then the bus is blocked again. My BlueSCSI is the desktop version, it's connected to the external SCSI port and is terminated. Driver is HDDRIVER 11.12. When using the same cable connection with RaSCSI there are no issues.
The boot log looks fine:
```BlueSCSI <-> SD - https://github.com/erichelgeson/BlueSCSI
VER: 1.1-20220917-USB-128MHz
DEBUG:0 SDFAT_FILE_TYPE:3
SdFat version: 2.2.0
Sd Format: exFAT
SPI speed: 50Mhz
SdFat Max FileName Length: 64
Initialized SD Card - let's go!
Sd MID:FE OID:4V
Sd Name:ASTC
Sd Date:10/2021
Sd Serial:196
 - CD50_tt.iso CDROM / 487593984bytes / 476166KiB / 465MiB
 MODE2:0 BlockSize:2048
 - HD10_512_ibm.hda HDD / 9139200000bytes / 8925000KiB / 8715MiB
ID:LUN0:
 0:----:
 1: 512:
 2:----:
 3:----:
 4:----:
 5:2048:
 6:----:
Finished initialization of SCSI Devices - Entering main loop.
```

uweseimet, 2022-11-03:
Now I tried the internal port, and this time it works. @fdanapfel You said you replaced your internal disk. Have you also tried BlueSCSI at the external port?

fdanapfel, 2022-11-03:
@uweseimet I've only tried the Desktop (50pin) version of BlueSCSI on the internal connector of the TT via a standard 50pin ribbon cable, without having any other SCSI devices attached, with the terminators of the BlueSCSI enabled (both jumpers present).

I've not tried it on the external connector since I don't have a fitting cable. 

But I also have a BlueSCSI with the 25-pin connector and this does not work on the external connector of the Atari TT03, but I suspect it is due to the fact that the TT does not supply TermPower on the SCSI connectors since the LEDs on the BlueSCSI don't even light up when the TT is powered on.

Using an external RaSCSI connected to the external SCSI port works fine however (also without having any other SCSI devices except for a Terminator on the 50 pin internal ribbon cable attached).

The SCSI terminators on the TT mainboard have been removed a long time ago as far as I can remember, otherwise using either the internal or the external SCSI connector wouldn't work.

Hope this helps.

uweseimet, 2022-11-03:
@fdanapfel Some kind of termination problem appears to be likely, and a blocked bus would be typical for that. @erichelgeson Maybe you are aware of similar issues?
@fdanapfel I think I have the required cables or adapters to externally connect BlueSCSi with a connection as short as possible. Maybe this will help. Using it internally is not my preferred setup because for testing purposes just switching between external devices is more convenient.

uweseimet, 2022-11-03:
@fdanapfel The shortest external connection I can use for the desktop version is about 40 cm. With this one there are still issues when loading files, but the devices are now properly detected and I get halfway through the boot process. Looks as if cable length is more crucial for BlueSCSI than it is for RaSCSI, indeed.

erichelgeson, 2022-11-04:
A few things to try with termination issues - measure 5v on the BlueSCSI - since we use passive termination from the bus usually if it's too low it wont be enough to pull up the termination resistors. You could try powering it via a USB power adapter to rule that out as well. I do use my BlueSCSI on a Mac with quite a long external SCSI cable. RaSCSI is always powered externally so that isn't as big of an issue.

uweseimet, 2022-11-04:
@erichelgeson Does it make any difference whether I power via USB or via the Berg connector? I used the latter when connecting BlueSCSI to the external port of my TT, but despite external power there were issues.

erichelgeson, 2022-11-04:
If they both supply 5v then it wont matter. I would just try both as it will quickly rule it in or out. I see many vintage machines have a low 5v line or share the same 5v on the termpower and berg.

uweseimet, 2022-11-04:
I am using a separate power supply for BlueSCSI only. I also tried a different power supply, but there was no change unfortunately.