erichelgeson/BlueSCSI-v1 #191: Support for LUNs other than 0

uweseimet, 2022-11-04:
At least for users of an Atari MegaSTE it would be great if BlueSCSI was supporting other LUNs than only LUN 0. The MegaSTE has a restricted internal SCSI port, which only supports a single SCSI ID/device, i.e. it cannot access more than one BlueSCSI image file. With support for several LUNs MegaSTE users could not just use BlueSCSI as a replacement for exactly one hard disk drive, but also for more hard disk drives or a CD-ROM drive.

erichelgeson, 2022-11-04:
We addressed a few issues with LUN handling a while back but never actually added the ability back (macs don't support LUNs so I have no way of testing). If you'd like to take a crack at it please do!

Site note you can have many SCSI ID's on one bluescsi - so you should be able to support multiple disks/cds on one bluescsi - just name them HD0..., HD1... etc

uweseimet, 2022-11-04:
@erichelgeson That's just the problem you have with the MegaSTEs controller: It does not support more than exactly one SCSI ID. It is configured with the ID that shall support, and any other ID is not accessible. That's why for this adapter the only work-around is to have several LUNs, which I tested successfully with RaSCSI.
I could run tests also for BlueSCSI (assuming that I would manage to update the firmware), but I'm afraid I won't be able to provide code.

mynameistroy, 2022-11-04:
I was looking at adding support for this as I have a couple of MegaSTE's in which to test.

uweseimet, 2022-11-05:
Having support for this would be great. I already had issues because with just one device transferring data to my MegaSTE is rather complicated. With RaSCSI I use a second drive image, which I prepare with Hatari, for instance. With two drives available I can then just copy the data I need from one drive/image to the other in the GEM desktop. With only a single drive I need to use a floppy disk for the data transfer, which is quite a pain.

mynameistroy, 2022-11-05:
Agreed, definitely a convenience feature for that situation. I'll try to get something going shortly for it.

mynameistroy, 2022-11-13:
There is now a PR with support for this feature specific to the MegaSTE situation so it will likely get pulled into the next release.

mynameistroy, 2022-11-14:
This is a very limited mode that simply allows the 7 available SCSI IDs to be re-mapped as LUNs on SCSI ID0 specifically for the MegaSTE. It was done to make life easier for MegaSTE users for the time being with respect to the minimal resources available on the F103. I hope in good time full LUN support will be present for those that need it once further refactoring is finished.

Robbizz, 2023-03-10:
Hi all, I don't know if this is the right place but I too am having trouble getting my two Atari MSTEs to see multiple drives. I bought 2 BlueSCSI model 50-Pin Desktop (v1.1a) based on a description I found here: https://github.com/erichelgeson/BlueSCSI/wiki/Advanced-Settings#mega-ste-mode-mste_mode
But it seems that it is only possible from HDDRIVER but honestly I should spend another 60 Euros and at the moment I don't like it. I tried with ICD (once the SD card was prepared according to the instructions found in the link above) but without success.

erichelgeson, 2023-03-10:
I was unaware that there was a paid driver in 2023 for a vintage computer. Until it works with a free driver I think it's best to remove the docs and put a note on the compatibility page to avoid confusion. Full disclose I've never used/seen a Atari computer, so relying on users reports.

uweseimet, 2023-03-10:
As the author of HDDRIVER I am not neutral, but fact is that in order to use multiple drives with the Atari and BlueSCSI you need a driver with support for multiple logical SCSI unit (LUNs). HDDRIVER is the only driver for the Atari with support for up to 8 LUNs.
Note, though, that I am having issues at least with the latest BlueSCSI firmware. See https://github.com/erichelgeson/BlueSCSI/issues/190 for details. In case you decide to purchase HDDRIVER first check whether the demo version works with your hardware.