erichelgeson/BlueSCSI-v1 #197: Bootable in a AT machine?

jensma-de, 2022-11-19:
Hey there!
I'm trying to get a bluescsi running in an old AT machine.
The installation of DOS5 works fine, partitioning is no problem and setting the primary partition active also works without a hitch.

Unfortunatly I don't seem to be able to boot from the bluescsi. Is that normal? Or is there a trick to it =)?

Cheers!

mynameistroy, 2022-11-20:
The ability to boot from a SCSI drive on a PC would be in the hands of the ROM on the SCSI card/chipset you are connecting to the BlueSCSI.

What is the hardware you are working with specifically?

jensma-de, 2022-11-20:
[image]
[image]

That's the machine :)
The SCSI controller is a NCR 53C400 and the BIOS for it resides in the eprom with the faded sticker "RT-BIOS".
Here's some info about it: https://forum.vcfed.org/index.php?threads/rancho-rt1000-8-bit-isa-scsi-controller-drivers-firmware.38039/

The PC seems to seek or wait endlessly for the bluescsi to boot, at least that's how I interpret the startup messages (or lack thereof)

jensma-de, 2022-11-20:
Fixed - burned a newer SCSI BIOS and works now =)

closed 2022-11-20 by jensma-de