erichelgeson/BlueSCSI-v1 #212: Can't seem to get BlueSCSI to talk to my NexSTstation

tomekwojcik, 2023-01-15:
Hello and greetings from Poland!

I recently got a BlueSCSI from a buddy who happened to have one lying around. I wanted to use it my NeXTstation Turbo Color, to replace SCSI2SD. Unfortunately, I can't seem to get it to work.

I lifted the image off of SD card I use in SCSI2SD. It's verified correct by running it in the emulator. The image is 4GB, 512 byte sector. I dropped it on an exFAT SD card under name `HD00_512_NS33.img`. The adapter detects the image just fine and assigns it to SCSI ID 0 (see `LOG.txt` below).

The machine doesn't want to talk to it at all. I get all sorts of craziness - from SCSI hangs (termination is on), through checksum errors, blk errors, all the way to kernel panics and read errors when booting. There's no pattern, sometimes the thing will detect the disk and try booting, sometimes it'll error out on SCSI detection, other times it'll just hang in the ROM Monitor.

I connected the adapter to a PC using Adaptec AVA-2904 SCSI controller and it just worked. The image got detected. I also played around with other images just to check if data transfers actually work and stuff. All's fine and dandy under both Linux and Windows XP.

Problematic machine: NeXTstation Turbo Color
Problematic host adapter: NCR 53C90A

What I tried:

- Using different SD cards and both FAT32 and exFAT.
- Assigning the image to different SCSI ID (0, 1, 2, 3).
- Downgrading the firmware - tried 20221203, 20220917 and 20220626.

Just wanted to let you know. If you happen to have any ideas what to do, I'll be happy to give them a go.

Anyways, thanks for a cool project. Even if it doesn't work in my NeXTstation, I'll have a reason to grab another SCSI-based machine to maybe try with it. I'm thinking an SGI O2 ;).

Have a good day (or night)! :)

DRAKE9800, 2023-01-15:
https://winworldpc.com/download/3fc5bdc2-bde2-80a2-533f-11c3a4c2a90f

For testing purposes please grab this 4.2 image and rename it under the BlueSCSI naming convention. 
"HD60_512 Next.hda" for example. 

This image is bootable on both my Next Cube 040 and Turbo 040 slab. 

Let me know if this still presents issues and we'll have to look elsewhere.

tomekwojcik, 2023-01-16:
Unfortunately, the OpenStep image didn't work either. On first boot, the machine threw a kernel panic immediately after the disk information was displayed. On the second boot, it didn't even manage to load the kernel from the root device (see screenshot below). On the third boot, it went into kernel panic loop until I powered it down.

Since my device is homemade, there's a chance it's faulty in some way which triggers the NCR controller on the NeXTstation. Is there a way to enable debugging to the LOG.txt?

[image]

erichelgeson, 2023-02-26:
> Is there a way to enable debugging to the LOG.txt

You will have to re-compile the firmware with DEBUG set to on - it's a compile time option. I've attached the latest for USB and non-USB bootloaders:

[BlueSCSI Debug.zip](https://github.com/erichelgeson/BlueSCSI/files/10834121/BlueSCSI.Debug.zip)