erichelgeson/BlueSCSI-v1 #215: flashing BlueSCSI firmware - Invalid DFU suffix signature

dragmar, 2023-01-25:
```
> Choose an option:
> 
> [1] Upgrade via USB.
>  - You must have BlueSCSI v1.1-20220404-USB or greater already flashed.
> [2] Prepare a blank BluePill or older BlueSCSI firmware for USB support.
>  - You must have a STLink v2.
> 
> [H] Help
> [P] Page Sysop
> [Q] Hang Up
> 
> Choice?2
> Flashing the generic_boot20_pc13.bin
> * Move BOOT0 to 1 (Jumper furthest from the RESET button on the BluePill)
> * Connect the ST-LinkV2 to your BlueSCSI and connect it to your computer.
> Press any key to continue . . .
> Starting flash...
> NOTE: Some BulePills can take up to 2 minutes to flash.
> xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:32)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> 0
> Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
> Warn : Transport "hla_swd" was already selected
> hla_swd
> Info : clock speed 1000 kHz
> Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
> Info : Target voltage: 3.110657
> Info : [stm32f1x.cpu] Cortex-M3 r2p0 processor detected
> Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
> Info : starting gdb server for stm32f1x.cpu on 3333
> Info : Listening on port 3333 for gdb connections
> target halted due to debug-request, current mode: Thread
> xPSR: 0x01000000 pc: 0x080000f0 msp: 0x20002800
> ** Programming Started **
> Info : device id = 0x10010414
> Info : flash size = 256kbytes
> ** Programming Finished **
> ** Verify Started **
> ** Verified OK **
> ** Resetting Target **
> shutdown command invoked
> Check output if successful (Windows...)
> * Move the BOOT0 Jumper back to position 0
> * Disconnect the BlueSCSI from the STLink
> Ready to flash BlueSCSI firmware.
> Press any key to continue . . .
> Flashing BlueSCSI-v1.1-20221203-STM32F1-USB.bin
> * Move BOOT1 to 1 (Jumper nearest the RESET button on the BluePill)
> * Connect via USB to the BluePill
> dfu-util 0.11
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2021 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> Warning: Invalid DFU suffix signature
> A valid DFU suffix will be required in a future dfu-util release
> Waiting for device, exit with ctrl-C
```
Bootloader loads into Bluescsi but FW does not. I've tried everything and I don't know what causes this problem.

In STM32 ST-LINK Utility v4.6.0 it verifies the bootloader file and everything is correct.

What am I doing wrong?

erichelgeson, 2023-01-25:
Could you please move this ticket over to the Updater repo? https://github.com/erichelgeson/BlueSCSI-Updater Thanks!

erichelgeson, 2023-02-09:
Closing as no response, please open in the correct repo if you continue to have an issue.