erichelgeson/BlueSCSI-v1 #216: Macintosh Classic compatibility with overclocked firmware

gingerbeardman, 2023-01-27:
- just a data point and to keep a record of this on GitHub
- I know there's not much point running this on a Classic 68K but one has to at least try :)

My Macintosh Classic setup shows "question mark drive" when running either 96MHz or 128MHz overlocked firmware. It's fine with the standard firmware.

BlueSCSI was flashed OK with no errors (though the DFU "Resetting USB to switch back to runtime mode" took a lot longer after flashing the overlocked firmwares) but the Mac won't see any drives through it. 

- I bought my BlueSCSI from flamelily UK if that helps 2022-04-11.
- Let me know if I can provide any additional logging.

gingerbeardman, 2023-01-27:
For interests sake...

SCSI2SD
[image]

BlueSCSI
[image]

erichelgeson, 2023-02-26:
Going to close this as the bluepill is likely just not able to run at the faster speed. OC is at your own risk and may or may not work - not much we can do from our side.

closed 2023-02-26 by erichelgeson