erichelgeson/BlueSCSI-v1 #227: Indicator extension LEDs

Elv1sP, 2023-03-31:
I cant for the world of me get the extended led lights to work. Got the ports and the resistors on the bluescsi but then, how do I get it to work, please explaine in laymansterms

erichelgeson, 2023-04-01:
Which version do you have? Are you using the original LED on an old machine or a new LED? The LED goes through a 330ohm resistor so depending on the LED you are using it may be dim (try to view it in a dark room to see if it's actually being lit or not). You'll need to calculate the right resistance for the LED you want to use. 330ohm usually is slightly dim for old LED's and fine for new LED's. Hope that helps. You can find some more info in the docs - https://github.com/erichelgeson/BlueSCSI/wiki/External-LED

Elv1sP, 2023-04-01:
Did not even think of that, I thought I was connecting the cords wrong. Just to clearify that, what is the correct way….and termination should only be used when alone in the chain right. If cd-drive or original HDD is in, No termination should be used?

erichelgeson, 2023-04-01:
Termination should be used on the last device on the chain, even if it's alone.

Elv1sP, 2023-04-01:
Yes I know but the bluescsi is first in the chain (cd-drive is last) my images are scsi 1-2-4 number 3 is cd-drive

erichelgeson, 2023-04-01:
SCSI ID numbers are different than where it is physically located on the chain and have no affect on one another. If your CD rom is last in the physical cable it will need to have it's termination on.

Elv1sP, 2023-04-01:
It has :)