erichelgeson/BlueSCSI-v1 #234: Desktop 1.1 3d print link is dead

chrisdaun, 2023-04-24:
The like to the 3d printable mount files for the 1.1 desktop version is dead: https://github.com/erichelgeson/BlueSCSI/blob/main/3D-Files/1.1/bluescsi11a_kerosmacmodsv2.stl.zip

erichelgeson, 2023-05-06:
Strange! Here it is for now:

[bluescsi11a_kerosmacmodsv2.stl.zip](https://github.com/erichelgeson/BlueSCSI/files/11412610/bluescsi11a_kerosmacmodsv2.stl.zip)

chrisdaun, 2023-05-06:
Thanks @erichelgeson

mitchkramez, 2023-05-08:
@erichelgeson this one doesn't seem to be what the bluescsi's came with, at least it doesn't work with a right angled connector - is there another version somewhere? I just went through all the branches and couldn't find a version that works with the right angled connector in any of them, but maybe i'm blind and totally missed it?

erichelgeson, 2023-06-19:
[bluescsi11a_kerosmacmodsv2.stl.zip](https://github.com/erichelgeson/BlueSCSI/files/11789479/bluescsi11a_kerosmacmodsv2.stl.zip)

Right angle compatible - sorry i thought that other zip was the same.

closed 2023-06-19 by erichelgeson