erichelgeson/BlueSCSI-v1 #239: Hard Disk shown 8 times

mfaridbadr, 2023-06-07:
HI,
I have 2 BlueSCSI ver.1.0c   
with same firmware (v1.1-20221203)
with same microSD card  and File 
the log on both   the same  result  

one working normal  and the other shown the blueSCSI as 8 times  (2 as Disk Drive , 6  as QUANTUM  )

[image]
[image]

What can I do to fix that

erichelgeson, 2023-06-08:
Are you positive both are on the same firmware? There was a bug in an earlier firmware that would show a device on each LUN the host requested, but it is resolved in the latest.

mfaridbadr, 2023-06-09:
Thanks Eric ,
You are correct ,