erichelgeson/BlueSCSI-v1 #255: Problem with large drive?

Grieverheart, 2023-10-15:
I have a Quadra 840AV for which I'm trying to prepare a drive on the BlueSCSI to store games. I made it 30GB using `dd` and formatted it on my Quadra using 'Drive Setup 1.7.3'. I installed many games on it (over 20GB) using Basilisk II, but when trying to use the disk on my Quadra, first of all it's really slow (it takes a couple of seconds to open the drive and scrolling also takes a few seconds), and the files seem to be corrupted. Whenever I try to run a game, I get an error that there's not enough memory, with zero available! Mounting the image again on Basilisk II, everything works fine, meaning that there is no corruption.

The version I'm using is `1.1-20211215`, which is quite old, but I'm waiting for my flasher to arrive so that I can try newer versions. Not sure if there are any possible fixes in later versions.

Not sure if the bug can be reproduced easily. I'll try creating a new large image and see if I get the same problems with a single file.

Grieverheart, 2023-10-16:
So I tried creating a large drive (20GB) and filled it with 200 folders and put 10 empty text files in each folder, but that failed to reproduce the problem. So either this happens when the drive is close to full, or Basilisk somehow corrupted it. It's weird that it seems fine in Basilisk, though.

erichelgeson, 2023-10-16:
The one thing with Basilisk II and some other emulators is that it doesn't really emulate scsi devices - it just searches for the first HFS looking partition and uses that. Glad you got it working and enjoy!

If you'd like an emulator that _does_ emulate SCSI (though a bit harder to use) is PCE (mac plus) or MAME.

closed 2023-10-16 by erichelgeson

Grieverheart, 2023-10-17:
Unfortunately I have not gotten the drive working yet properly using BlueSCSI. Do note that the drive is mounting, it's just that most binaries do not work and the icons look corrupted, like in the picture below. 

I understand that Basilisk II doesn't emulate SCSI, otherwise it would probably not be as fast as it is. That's also the reason I used Basilisk to install everything. Un-Stuffing on the Macintosh takes considerably more time.

I built a Python parser for inspecting the apple partition maps etc., but I'm not seeing anything weird there. Do you have any tips where things could be going wrong? Could the resource forks be pointing in an area not accessible by the actual machine?

Also, is installing everything on the Quadra instead of Basilisk the safest bet? I spent a whole day doing installations, on the Quadra it will probably take even longer, so I don't want to have to redo that.
[image]

Grieverheart, 2023-10-18:
Finally received my STlinkV2, so I flashed the BlueSCSI to the latest version. To my surprise, the files seem to be fine now, and games launch fine!

erichelgeson, 2023-10-20:
ahh! I misread your second message as it was only in BII that it was corrupted, glad the update got it going!