erichelgeson/BlueSCSI-v1 #260: BlueSCSi V2 returns wrong SCSI error codes for a non-existing LUN

uweseimet, 2024-01-15:
Tested with the current firmware: When a SCSI driver sends REQUEST SENSE for LUN 7 to a BlueSCSI devices configured with ID 0 and LUN 0 (i.e. no LUN specified in the image filename) BlueSCSI returns Sense Key $00, ASC, $00.
The SCSI-2 specification says this shall be Sense Key $05, ASC $25:
```
7.5.3 Selection of an invalid logical unit

The target's response to selection of a logical unit that is not valid is
described in the following paragraphs.

The logical unit may not be valid because:
   a)     the target does not support the logical unit (e.g. some targets
          support only one peripheral device).  In response to an INQUIRY
          command, the target shall return the INQUIRY data with the
          peripheral qualifier set to the value required in 8.2.5.1.  In
          response to any other command except REQUEST SENSE, the target shall
          terminate the command with CHECK CONDITION status.  In response to
          a REQUEST SENSE command, the target shall return sense data.  The
          sense key shall be set to ILLEGAL REQUEST and the additional sense
          code shall be set to LOGICAL UNIT NOT SUPPORTED.
```

erichelgeson, 2024-01-15:
Hi - this repo is for BlueSCSI v1 - you can find v2 here https://github.com/BlueSCSI/BlueSCSI-v2 - I can not transfer the issue as it's across a different github org.

uweseimet, 2024-01-15:
@erichelgeson I see, I'm sorry for that. I used an old link in my bookmarks. I will close this ticket and re-create it.

closed 2024-01-15 by uweseimet