erichelgeson/BlueSCSI-v1 #264: BlueSCSI v1 firmware update needs a step-by-step set of instructions

someyob, 2024-04-19:
1.  How to use QMK for this?
2.  Which firmware, there's several to choose from?
3.  Am I plugging in to the Bluepill, or using a ST-Link?
4.  Do I need to change the jumpers on the Bluepill?

tiff-lee, 2024-07-18:
Did you figure this out in the end? I took a bunch of pictures showing the process using the ST-Link and the ST32 programming software for the blue pills but I hadn't got around to knocking a guide up yet. If you still need it happy to do something.

tiff-lee, 2024-07-29:
Ok sorry for the delay, i've just put something together its a little rough around the edges but pretty much anyone should be able to follow this even with little prior knowledge.
Note that this is using a clone ST-LINK V2 and was flashing the bluepill for the bluescsi v1.1.

[Flashing Guide.pdf](https://github.com/user-attachments/files/16411665/Flashing.Guide.pdf)

someyob, 2024-07-29:
Lucky day, today was my day to do mac stuff.

I ran through the procedure, no issues whatsoever.  Only one ambiguous section:  When connecting the jumpers between the ST-Link and the BluePill, I chose to do it _without_ the ST-Link connected to the USB.   Once finished connecting, I plugged in the ST-Link to USB and carried on per page 3.  I figured it was safest to do it unpowered; maybe it's irrelevant.

Thanks very much for doing this, all booting fine.

tiff-lee, 2024-07-29:
Excellent, yeah that text was pulled directly from the wiki there is no real danger either way. Main thing is you got it sorted which is good.

talljon, 2025-02-16:
tiff-lee your Flashing Guide.pdf is very well done and easy to follow.  

Thank you for posting.