erichelgeson/BlueSCSI-v1 #276: Feature Request: make the shared folder available on SMB

Talismanian, 2024-10-31:
* The `shared` folder is extremely useful in letting us quickly transfer files from modern machines.
* However, getting files into the folder requires disconnecting the BlueSCSI to access the SD card.
* Emulation platforms like RetroPie usually allow quick transfer of ROMs and files via exposing specific folders over SMB.
* Exposing the `shared` folder over SMB could allow file transfers to be even faster.
* I don’t have an opinion on whether the whole SD card should be accessible on SMB too.

I’ve been enjoying using BlueSCSI with my Mac Plus. Even though I can collect it to wifi, I’m happy to keep letting it run as a game/typing box. The Transfer Utility is a godsend, but I wish there was a way to make it more seamless.

erichelgeson, 2024-10-31:
Hi I think you meant to open this on the BlueSCSI v2 repo https://github.com/BlueSCSI/BlueSCSI-v2 - but the quick answer is there is not enough cpu time or flash size in the Pico1-W to implement this - and since you're already on wifi you could just use Fetch(FTP) or any other network transfer to grab files from your modern machines to your Plus - I use Netatalk to share between my NAS and my Mac Plus. 

HTH - please open a discussion on the v2 repo if you have more questions, Thanks!

closed 2024-10-31 by erichelgeson

Talismanian, 2024-11-04:
Thank you for that and apologies